Xinyuan-LilyGO/LilyGo-LoRa-Series #194: Can't upload sketches to T3 S3

TigoTas, 2024-08-13:
Hi,

I have a similar issue to the one described in #121, but the proposed fix doesn't work for me.

I do get the board to be recognized and the COM port to appear, but when trying to upload, I get a timeout exception.

I purchased two T3 S3. The second board works fine.

Any help would be appreciated.

[image]
[image]
[image]

lewisxhe, 2024-08-14:
Press and hold the BOOT button, then insert the USB, remember not to connect the battery. It seems that you have not entered the download mode.

TigoTas, 2024-08-14:
Thank you, @lewisxhe. It worked like a charm.

Not exactly as I expected, but worked nonetheless.

The behaviour I observed:

- Green LED beside blinking blue LED turned on, upon performing actions according to your response;

> Press and hold the BOOT button, then insert the USB, remember not to connect the battery. It seems that you have not entered the download mode.

- COM port 19 was identified, instead of 13;
- Upload failed just the same;
- Re-did the same steps;
- Green LED no longer on;
- COM 13 identified;
- Upload succeeded;