Xinyuan-LilyGO/LilyGo-LoRa-Series #195: SD_MSC and board re-start

TigoTas, 2024-08-15:
Hi,

I am having issues trying to run the SD-FAT version for exposing the SD card as Mass Storage.

**More of a weird behaviour.**

I am using the[ TTGO T3-S3](https://www.lilygo.cc/products/t3s3-v1-0) (ESP32S3FH4R2) and coding with Arduino IDE 2.3.2 and the latest Espressif framework.

I have made few modifications to the original code (SD card is started in a HSPI interface).

It compiles and uploads without a hitch, but then the weird part begins. What I observed:

- After upload and board re-start, nothing comes up in the serial monitor;
- Also, after upload and board re-start, the MSC doesn't mount;
- If I unplug the board and then plug it back, the MSC mounts normally.

Now...the board has a RESET switch, but **the  only thing that works is physicaly pulling the cable off and plugging it back in**.

I am not sure whether there are code modifications I can try, or if this is purely board related...

Perhaps @lewisxhe could weight in here.

It would be nice to have a workaround for this. My intended application doesn't allow for someone to pull the plug and plug it back when needed.

Thanks in advance.

`#include <SPI.h>
#include <SdFat.h>
#include <Adafruit_TinyUSB.h>

#define SDCARD_MOSI                 11
#define SDCARD_MISO                 2
#define SDCARD_SCLK                 14
#define SDCARD_CS                   13

SPIClass SDSPI(HSPI);

SdFat sd;

SdFile root;
SdFile file;

Adafruit_USBD_MSC usb_msc;

bool fs_changed;

void setup()
{
  Serial.begin(115200);

  // Set disk vendor id, product id and revision with string up to 8, 16, 4 characters respectively
  usb_msc.setID("Adafruit", "SD Card", "1.0");

  // Set read write callback
  usb_msc.setReadWriteCallback(msc_read_cb, msc_write_cb, msc_flush_cb);

  // Still initialize MSC but tell usb stack that MSC is not ready to read/write
  // If we don't initialize, board will be enumerated as CDC only
  usb_msc.setUnitReady(false);
  usb_msc.begin();

  //SPI.begin(SDCARD_SCLK, SDCARD_MISO, SDCARD_MOSI, SDCARD_CS);
  SDSPI.begin(SDCARD_SCLK, SDCARD_MISO, SDCARD_MOSI, SDCARD_CS);
  SdSpiConfig config(SDCARD_CS, DEDICATED_SPI, SD_SCK_MHZ(25), &SDSPI);
  
  //if (!sd.begin(SDCARD_CS))
  if (!sd.begin(config))
  {
    Serial.println("SD FAILED!");

    while (1)
      delay(1);
  }
  else
  {
    Serial.println("SD OK!");
  }

  // Size in blocks (512 bytes)
  #if SD_FAT_VERSION >= 20000
    uint32_t block_count = sd.card()->sectorCount();
  #else
    uint32_t block_count = sd.card()->cardSize();
  #endif

  Serial.print("Volume size (MB):  ");
  Serial.println((block_count/2) / 1024);

  usb_msc.setCapacity(block_count, 512);
  usb_msc.setUnitReady(true);

  fs_changed = true;
}

void loop()
{
  if ( fs_changed )
  {
    root.open("/");
    Serial.println("SD contents:");

    // Open next file in root.
    // Warning, openNext starts at the current directory position
    // so a rewind of the directory may be required.
    while ( file.openNext(&root, O_RDONLY) )
    {
      file.printFileSize(&Serial);
      Serial.write(' ');
      file.printName(&Serial);
      
      if ( file.isDir() )
      {
        // Indicate a directory.
        Serial.write('/');
      }

      Serial.println();
      file.close();
    }

    root.close();

    Serial.println();

    fs_changed = false;
    delay(500);
  }
}

// Callback invoked when received READ10 command.
// Copy disk's data to buffer (up to bufsize) and
// return number of copied bytes (must be multiple of block size)
int32_t msc_read_cb (uint32_t lba, void* buffer, uint32_t bufsize)
{
  bool rc;

  #if SD_FAT_VERSION >= 20000
    rc = sd.card()->readSectors(lba, (uint8_t*) buffer, bufsize/512);
  #else
    rc = sd.card()->readBlocks(lba, (uint8_t*) buffer, bufsize/512);
  #endif

  return rc ? bufsize : -1;
}

// Callback invoked when received WRITE10 command.
// Process data in buffer to disk's storage and 
// return number of written bytes (must be multiple of block size)
int32_t msc_write_cb (uint32_t lba, uint8_t* buffer, uint32_t bufsize)
{
  bool rc;

  digitalWrite(LED_BUILTIN, HIGH);

  #if SD_FAT_VERSION >= 20000
    rc = sd.card()->writeSectors(lba, buffer, bufsize/512);
  #else
    rc = sd.card()->writeBlocks(lba, buffer, bufsize/512);
  #endif

  return rc ? bufsize : -1;
}

// Callback invoked when WRITE10 command is completed (status received and accepted by host).
// used to flush any pending cache.
void msc_flush_cb (void)
{
  #if SD_FAT_VERSION >= 20000
    sd.card()->syncDevice();
  #else
    sd.card()->syncBlocks();
  #endif

  // clear file system's cache to force refresh
  //sd.cacheClear();

  fs_changed = true;

  digitalWrite(LED_BUILTIN, LOW);
}`

lewisxhe, 2024-08-16:
I used espressif's USBMSC to test that everything worked fine. You can try it

TigoTas, 2024-08-16:
Yes. That sketched DOES work.

But if I understand properly, the mentioned code sets a FAT table and exposes it as MSC through USB.

What if I want the exposed volume to be my SD card, as in the [SD-FAT example from Adafruit](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/MassStorage/msc_sdfat)? This is the code I mentioned that only works when you power down the board and back on again, but not when pressing reset.

Is it possible to use the "USBMSC.h" lib to expose an SD card as a MSC? Can you, perhaps, point me to any code example?

lewisxhe, 2024-08-16:
Did you miss my quote?
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/commit/e865cad8243db266d7759faf724719dd53a6be2a

TigoTas, 2024-08-16:
I am really sorry. I did missed it.

Thanks. That works perfectly.

closed 2024-08-16 by TigoTas