Xinyuan-LilyGO/LilyGo-LoRa-Series #201: Compile an example code for T-Beam v1.2

sirfonzie, 2024-09-27:
I am having issues even to compile the example code in Arduino as when I execute the code, it goes into a infinite watchdog reset loop. 
I followed the setup for ESP32 Dev Module. I cant figure out where I am making a mistake.

lewisxhe, 2024-09-27:
Arduino IDE board settings, running examples, tell me respectively

sirfonzie, 2024-09-27:
[image]

I am using the T-Beam V1.2 (SX1262), thus I uncommented (in utilities.h file):
#define T3_S3_V1_2_SX1262

Running this example: https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/Transmit_Interrupt

When compiled and uploaded, I get this error:

-----------------------------------
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
setupBoards
-----------------------------------
Reset reason: In case of deep sleep, reset was not caused by exit from deep sleep
PSRAM is enable! PSRAM: 4.00MB
Flash:4 MB
Flash speed:80 M
Model:ESP32-D0WDQ6-V3
Chip Revision:45
Freq:240 MHZ
SDK Ver:v5.1.4-828-g33fbade6b8-dirty
DATE:Sep 27 2024
TIME:17:30:07
EFUSE MAC: 606DF6E0F908
-----------------------------------

lewisxhe, 2024-09-27:
Try lowering the Arduino ESP32 version to 2.0.14

[image]

moshe-braner, 2024-09-30:
On the other hand need at least 2.0.15 to get the fixes for Serial output via USB if compiling for the ESP32-S3.  So cannot use the same environment to compile for both devices?  What is the bug in 2.0.15+ regarding the T-Beam?

lewisxhe, 2024-10-01:
It should work with 2.0.15, but I don't have the board now, so I can't test it. I will test it later. Theoretically, it can also run with 3.0 or above, but if there is a problem, downgrade first to see if the same problem occurs.

sem-feldoe, 2024-10-19:
Hello i got mine today ( 866MHz )
Used #define T3_S3_V1_2_SX1276 and got the same behavior:

rst:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
setupBoards
-----------------------------------
Reset reason: In case of deep sleep, reset was not caused by exit from deep sleep
PSRAM is disable!
Flash:4 MB
Flash speed:80 M
Model:ESP32-D0WDQ6-V3
Chip Revision:45
Freq:240 MHZ
SDK Ver:v5.1.4-828-g33fbade6b8-dirty
DATE:Oct 19 2024
TIME:13:30:10
EFUSE MAC: 64FB726CDDA0

I downgraded but then it fails to compile.
@sirfonzie Did you find a way to make it work?
Thanks

lewisxhe, 2024-10-21:
Can't compile? Please take a screenshot or provide a log of the compilation error

lewisxhe, 2024-10-21:
[image]
[image]
[image]
Just tested the latest 3.0.5 ArduinoLoRa and RadioLib can work properly

sem-feldoe, 2024-10-21:
Hello @lewisxhe ,
I did a lot of tests, i can make it working if i use
#define T3_V1_3_SX1276
and i commented "beginWiFi();" that makes the board crash from setupBoards().
It should be nice to have a table for matching label on board and the define to use in utilities.h
i have a T-BEAM-AXP2101-V1.2, it's quite hard to guess that we need to use T3_V1_3_SX1276 to make it works!
Now that i can make a simple sketch works, i will try more things,
thanks lewisxhe!

lewisxhe, 2024-10-21:
beginWiFi() requires a lot of power. If you add a battery, it won't crash. If you don't need it, you can remove it first.

sem-feldoe, 2024-10-21:
I tried a more powerfull usb port and i can use Wifi and connect to a router etc... It's nice! thanks.

closed 2024-12-05 by github-actions[bot]