Xinyuan-LilyGO/LilyGo-LoRa-Series #205: Programming on LILYGO T-Beam Supreme module using Arduino IDE

Tanay7, 2024-11-12:
Product Link:
https://www.lilygo.cc/products/t-beam-supreme-meshtastic?variant=43067943911605

Description:

How can I get the LILYGO T-Beam Supreme (SOC: ESP32-S3FN8) to work with custom code, such as displaying "Hello" on its OLED, generating random numbers, and reading from an externally attached AS7341 spectral sensor, using the Arduino IDE? 
I understand this isn't the standard setup for the Meshtastic firmware, but I need to use the board for other applications as well. 
Previously, I successfully used the LILYGO® TTGO LoRa32 V2.1_1.6 (selected via Tools -> Board -> ESP32 -> TTGO LoRA32-OLED) for various projects, including a random number generator/transmitter and an MMC5603 magnetometer Lora transmitter/receiver. 
I am able to upload similar code to the T-Beam Supreme by selecting 'ESP32S3 Dev Module' in Tools -> Board, as there is no specific option for the T-Beam Supreme in the Boards Manager. 
However, after uploading, neither the embedded OLED nor the LED activates, and there is no output in the Serial Monitor. Could you please help me troubleshoot this issue? Additionally, it would be great if you could add a T-Beam Supreme board option in the Arduino IDE's Boards Manager, along with some ready-made sketches—for example, to retrieve and display data from the embedded BME280/QMI8658 sensors on its OLED. 

For reference, here are some sample codes that worked with the LILYGO® TTGO LoRa32 V2.1_1.6: 
https://github.com/Tanay7/IoT

lewisxhe, 2024-11-13:
Most of the examples in the repository can be run on T-Beam Supreme. The board is ESP32S3 DEV MODULE, as described in the README
[image]
OLED display is listed in the home page example
[image]
Please note that the board has a PMU, and the PMU power output must be turned on before the peripherals can be powered. Each example includes LoRaBoards.cpp, which initializes each setting of the board
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/95553f48e7b42b345662defe11376150f40119db/examples/OLED/SH1106FontUsage/LoRaBoards.cpp#L554
Arduino IDE users must first uncheck the pin definition of the corresponding board in each example above utilities.h before use
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/95553f48e7b42b345662defe11376150f40119db/examples/OLED/SH1106FontUsage/utilities.h#L27

lewisxhe, 2024-11-14:
> Thanks a lot, seems to work. Just one suggestion: Could you please rename 'isBatChagerStartIrq' function to 'isBatChargeStartIrq' function and 'isBatChagerDoneIrq' function to 'isBatChargeDoneIrq' function in all LoRaBoards.cpp files as it is causing compilation error?
This is caused by new API changes. You are not using the copy in the lib directory, or you updated XPowersLib through Arduino IDE, which caused the compilation error. I have updated the copy and examples in the lib directory.

lewisxhe, 2024-11-14:
> setupBoards

setupBoards is the initialization method function, including PMU initialization. You can change it according to your needs, but generally, you don’t need to modify it because it will initialize all peripherals of the board.

Tanay7, 2024-11-16:
> > setupBoards
> 
> setupBoards is the initialization method function, including PMU initialization. You can change it according to your needs, but generally, you don’t need to modify it because it will initialize all peripherals of the board.

Thanks, you may now close this issue.

closed 2024-11-18 by lewisxhe