Xinyuan-LilyGO/LilyGo-LoRa-Series #208: T-Beam S3 - AXP2101 Power Button Short Press Detection

Wellco, 2024-11-24:
Hello LilyGO Support,

I'm working with your T-Beam S3 board and trying to implement short press detection for the power button (while maintaining the default long-press power off functionality).

Despite trying multiple approaches from various implementations (including checking your examples), I cannot get reliable short press detection working. Here's what I've tried:

1. Direct register access to AXP2101:
   - Configuring POWER_KEY_PARAM (0x27)
   - Setting up interrupts via INTEN registers
   - Reading INTSTS registers

2. Current behavior:
   - I2C communication works (can read/write registers)
   - Power button works for power on/off
   - But short press detection either:
     * Shows constant interrupt flags (0x8F in INTSTS2)
     * Or doesn't detect presses at all

Could you please:
1. Confirm if short press detection is possible while maintaining long-press power off?
2. If yes, provide guidance on the correct register configuration?
3. Share any example code for this specific functionality?

Board: T-Beam SUPREME S3 v3
Environment: Arduino IDE 2.3.3
Power Management: AXP2101

Thank you for your help.

Wellco, 2024-11-24:
Reading my original post I wrote "cannot get reliable short press detection working". I should have said I cannot get any power button presses to respond at all

lewisxhe, 2024-11-25:
Run this example to see if it works properly.

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/PMU

Wellco, 2024-11-25:
Thank you so much for the example code Sir!!

I have my power button code working now.

Many thanks

closed 2024-11-25 by Wellco