Xinyuan-LilyGO/LilyGo-LoRa-Series #209: LoRa32 V2.1_1.6 gets really hot, but is working fine

thatSFguy, 2024-12-01:
I have several LoRa32 V2.1_1.6 devices, but only this particular one gets really hot.  The chip next to the antenna is so hot you have to let go before it burns you.  The firmware that I'm running is Meshtastic.  By all accounts the node is working fine, but I'm worried that it might burn up, since it is really hot.  

I've seen some similar closed issues, but none of them provide a real answer to my issue. 

Thanks!

lewisxhe, 2024-12-02:
I also took one to test, I chose the T-LoRa-v2.1-1.6 variant, version 2.5.15.79, then flashed into the device, connected to the mesh app, selected the region EU868, and then I touched the board area with my hand, and it didn't get hot. Is there anything else I need to do to test it?

Do you have a USB ammeter? You can measure the current of the USB power supply, or the battery power supply current to simply test if there is a problem

lyusupov, 2024-12-02:
> I have several LoRa32 V2.1_1.6 devices, but only this particular one gets really hot. 
> The chip next to the antenna is so hot you have to let go before it burns you.

Which one ?

[image]

If I would have such an issue - I would take a lens or a microscope and inspect all the ICs I can see on the PCB by criteria of are there any neighbor pins short by "solder bridge" while in production.

There are plenty of pins that are not in use by a particular firmware but they may have an unwanted connection to GND or power supply chains. This unwanted connection may cause high current drain which will result in increased heat dissipated by DC-DC or linear regulators ICs.

thatSFguy, 2024-12-02:
[image]
The circled chips get hot.  The one closest to the antenna get hot within seconds of powering up.  Sorry I don't have an usbameter. I have a total of 10 of these, and like I said this is the only one.  My laptop usually complaints if the power draw is too high, but not with this device.  Really strange.

lyusupov, 2024-12-02:
Both of these ICs (U8 and U9) are ME6211 LDOs 
They produce 3.3V voltage from 5V source.
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/T3_V1.6.1.pdf

 4 years ago I've received a free sample board from LilyGO.  ( **Thank you Lewis !** )
It was  T3 board with HPD16A radio module.

The T3 board was heating up every time when onboard radio (HPDTeK module) was active.

I've removed the shield and saw this

[image]

I've soldered the capacitor back to it's place - the board no longer heats up since then.

thatSFguy, 2024-12-03:
Thanks!  I will contact Lillygo Official Store via Aliexpress.

lyusupov, 2024-12-03:
@thatSFguy 

[image]

closed 2024-12-03 by lewisxhe