Xinyuan-LilyGO/LilyGo-LoRa-Series #210: L76K GPS locking up

Wellco, 2024-12-01:
I have a T-Beam Supreme version 3.

I am writing  an application to create an autonomous GPS data logger that will run unattended in my vehicle. testing so far has been good apart from one issue where the GPS appears to lock up when recording. If go out with the T-Beam it records for around 20 mins, then it hangs for 10 mins before it recovers;  and it more-or-less repeats this behaviour for the remainder of my travels. The satellite count is good, as is HDOP (0.9 - 0.8 much or the time). I even tried the device walking and the same happens. I'm on open countryside. 

Here's a code snippet if it helps..

    // L76K-specific configuration commands
    const char* gpsCommands[] = {
        "$PMTK251,115200*1F",     // Set baud rate to 115200
        "$PMTK314,0,1,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0*28",  // Enable RMC & GGA
        "$PMTK220,200*2C",        // Set update rate to 5Hz (200ms)
        "$PMTK397,0*23",          // Disable nav speed threshold
        "$PMTK386,0*23",          // Disable position pinning
        "$PMTK447,1*37",          // Enable EASY function
        "$PMTK225,0*2B",          // Disable periodic mode
        "$PMTK869,1,1*35"         // Enable EASY function
        "$PCAS04,7*33"            // Set constellation mode to GPS + BeiDou + GLONASS

Is anything strange here?

I'm struggling to find documentation for the L76K, so much of this is cobbled together from other code examples I could find.

I am writing a position in my GPX file every second. Do you thing the is overwhelming the T-Beam?

Many thanks

Wellco, 2024-12-02:
Many thanks for the documentation, is there a version available in English? It's interesting that you say the PMTK fields are invalid, since at least the one that produces the NMEA sentences does actually work. However I will convert my code to PCAS, but would really appreciate an version English of the documentation if you have one..

lewisxhe, 2024-12-02:
There is no L76K on the English website, it can only be found on the Chinese website, so I don’t have an [English version](https://www.quectel.com/product-selector/?_sft_product_cat=gnss-modules-standalone)

closed 2025-01-17 by github-actions[bot]