Xinyuan-LilyGO/LilyGo-LoRa-Series #212: T-Beam v1.2 L76K boot problem

wvthoog, 2024-12-22:
Hi all,

I've bought a LilyGO T-Beam V1.2 (AXP2101 / LoRa 868MHz / NEO-M8N GNSS / ESP32) and am trying to get the LoRaWAN RadioLib example to work. 

In platformio.ini i've set

```
default_envs = T_BEAM_SX1276
src_dir = examples/LoRaWAN/RadioLib_OTAA
```

and of course changed RADIOLIB_LORAWAN_DEV_EUI, RADIOLIB_LORAWAN_APP_KEY and RADIOLIB_LORAWAN_NWK_KEY

The problem is that (when i look at serial monitor) it's trying to init the L76K GPS chip and than 'hangs'. As in it doesn't go further

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
setupBoards
-----------------------------------
Reset reason: In case of deep sleep, reset was not caused by exit from deep sleep
PSRAM is enable! PSRAM: 4.00MB
Flash:4 MB
Flash speed:40 M
Model:ESP32-D0WDQ6-V3
Chip Revision:3
Freq:240 MHZ
SDK Ver:v4.4.7-dirty
DATE:Dec 20 2024
TIME:15:08:39
EFUSE MAC: 34:98:7A:6C:08:98
-----------------------------------
Scan Wire...
I2C Devices scanning
        Find AXP192/AXP2101 PMU!
Scan devices done.

AXP2101 PMU init succeeded, using AXP2101 PMU
=========================================
DC1  : +   Voltage: 3300 mV
DC2  : -   Voltage: 0900 mV
DC3  : -   Voltage: 0900 mV
DC4  : -   Voltage: 1100 mV
DC5  : -   Voltage: 1200 mV
ALDO1: -   Voltage: 1800 mV
ALDO2: +   Voltage: 3300 mV
ALDO3: +   Voltage: 3300 mV
ALDO4: -   Voltage: 2900 mV
BLDO1: -   Voltage: 1800 mV 
BLDO2: -   Voltage: 2800 mV
=========================================
PowerKeyPressOffTime:4 Second
Warning: Failed to find Display at 0x3C address
Try to init L76K . Wait stop ..
```

But when i press the reset (reboot) button a few time it does boot up fully and joins the LoRaWAN network

```
AXP2101 PMU init succeeded, using AXP2101 PMU
=========================================
DC1  : +   Voltage: 3300 mV
DC2  : -   Voltage: 0900 mV
DC3  : -   Voltage: 0900 mV
DC4  : -   Voltage: 1100 mV
DC5  : -   Voltage: 1200 mV
ALDO1: -   Voltage: 1800 mV
ALDO2: +   Voltage: 3300 mV
ALDO3: +   Voltage: 3300 mV 
ALDO4: -   Voltage: 2900 mV
BLDO1: -   Voltage: 1800 mV
BLDO2: -   Voltage: 2800 mV
=========================================
PowerKeyPressOffTime:4 Second
Warning: Failed to find Display at 0x3C address
Try to init L76K . Wait stop .
Try to init L76K . Wait stop ..
Try to init L76K . Wait stop ..
Update baudrate : 9600
Get ack successes!
Get ack successes!
Get ack successes!
Get ack successes!
UBlox GNSS init succeeded, using UBlox GNSS Module

init done .

Setup ... 
Initialise the radio
Initialise the radio
Restoring session buffer failed - RADIOLIB_ERR_NETWORK_NOT_JOINED (-1101)
Join ('login') to the LoRaWAN Network
Saving nonces to flash
[LoRaWAN] DevAddr: <replaced>
Ready!

Normal boot!
Sending uplink
Received a downlink
<MAC commands only>
[LoRaWAN] RSSI:         -42.00 dBm
[LoRaWAN] SNR:          7.75 dB
[LoRaWAN] Event information:
[LoRaWAN] Confirmed:    0
[LoRaWAN] Confirming:   0
[LoRaWAN] Datarate:     5
[LoRaWAN] Frequency:    868.300 MHz
[LoRaWAN] Frame count:  0
[LoRaWAN] Port:         0
[LoRaWAN] Time-on-air:  52 ms
[LoRaWAN] Rx window:    1
[LoRaWAN] Next uplink in 300 seconds
```

So my question is why does it hang at L76K ? (i didn't know the T-Beam v1.2 even had a L76K GPS chip, thought it had a Ublox GPS chip) and how can i let it boot up consistently 

Many thanks

lewisxhe, 2024-12-23:
T-Beam v1.2 does not have L76K, so just skip this part of the code

wvthoog, 2024-12-23:
Yep, it works. Edited LoRaBoards.cpp like this

```
#ifdef HAS_GPS
uint32_t baudrate[] = {9600, 19200, 38400, 57600, 115200, 230400, 460800, 921600, 4800};
bool find_gps = false;

for (int i = 0; i < sizeof(baudrate) / sizeof(baudrate[0]); ++i) {
    Serial.printf("Update baudrate : %u\n", baudrate[i]);
    SerialGPS.updateBaudRate(baudrate[i]);

    if (recoveryGPS()) {
        Serial.println("UBlox GNSS init succeeded, using UBlox GNSS Module\n");
        gps_model = "UBlox";
        find_gps = true;
        break; 
    }
}

if (find_gps) {
    deviceOnline |= GPS_ONLINE;
}

#ifdef T_BEAM_S3_SUPREME
    enable_slow_clock();
#endif

#endif
    Serial.println("init done . ");
}
```

closed 2024-12-23 by wvthoog