Xinyuan-LilyGO/LilyGo-LoRa-Series #216: GNSS button battery and PMU on T-Beam

moshe-braner, 2025-01-21:
The old code in SoftRF ESP32.cpp, using the old AXP library, ran this code:
     PMU->setButtonBatteryChargeVoltage(3100); // GNSS battery

But since I moved from the old AXP library to the new xPowersLib library, I had to comment out that line, since it is not supported by the xPowersLib.  I am using the "API" part of the library.  I tried this instead:
    PMU->enableButtonBatteryCharge();
but that would not compile either.

So I am not sure whether that button battery is being charged now?  After disconnecting power from the T-Beam for a while, the GNSS module (Ublox 6) often loses the leap seconds information, which takes up to 12 minutes to get from the GPS satellites.  I need the leap seconds for the exact UTC time since the encryption key for the radio messages is based on that.

There are two parts to this question:

*  is that button battery being charged while the T-Beam is running (the GNSS module is powered)?  Perhaps the GNSS charges that battery directly even if the separate wire from the PMU is not powered?
 
* is that button battery being charged while the T-Beam is "turned off", meaning the ESP32 has been put into deep sleep and the PMU has been told to stop powering the GNSS module.  This may depend on xPowersLib functions called.

Is there a way in xPowersLib to make sure that battery is charged?  This of course is only possible with an 18650 in the T-Beam's battery holder.  When I wrote above "after disconnecting power from the T-Beam for a while, the GNSS module often loses the leap seconds" I mean without an 18650, but sometimes this happens even with an 18650.

lewisxhe, 2025-01-22:
Turning off the PMU will not keep the RTC powered. The button battery has a very small capacity and can only last for a few minutes. The easiest way is to measure the voltage of the button battery, which is only valid when the PMU is turned on.
For information on turning on button battery charging, see here
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/169ead79b96d7ac4320c53fbb2bd91f2639648c1/examples/ArduinoLoRa/LoRaSender/LoRaBoards.cpp#L150

moshe-braner, 2025-01-22:
Thanks.  So it looks like this is the new PMU code to enable the button-battery charge:
        // GNSS RTC PowerVDD 3300mV
        PMU->setPowerChannelVoltage(XPOWERS_VBACKUP, 3300);
        PMU->enablePowerOutput(XPOWERS_VBACKUP);
(Is 3300mV the best value?)

And this code would disable that charge:
      PMU->disablePowerOutput(XPOWERS_VBACKUP);
but if I want the big 18650 cell to maintain the GNSS data should NOT do that (while otherwise de-powering the GNSS and putting the ESP32 into deep sleep)?

lewisxhe, 2025-01-22:
1.3300mV is the correct setting voltage
2. XPOWERS_VBACKUP should not be turned off in deep sleep, which will cause the GPS RTC power domain to power down and lose the saved ephemeris

closed 2025-03-09 by github-actions[bot]