Xinyuan-LilyGO/LilyGo-LoRa-Series #217: T beam SUPEREME

DanielPerezGuzman22, 2025-01-22:
Hello good morning. I'm trying to use the Deep Sleep features with a T Beam Supreme S3 but every time it executes an esp_deep_sleep_start(); Instead of going into sleep mode, the card reboots and starts running the program again. Could you tell me an example that works on this card. Thank you so much.

lewisxhe, 2025-01-23:
Which example do you use? Have you tested the built-in sleep example?

[image]

DanielPerezGuzman22, 2025-01-23:
Thank you very much for your help now it works properly. Now I will try to integrate it into RadioLib functions. Greetings

DanielPerezGuzman22, 2025-01-24:
Hello, I continue with the problem of restarting my T Beam Supreme S3. I ran the code you suggested and if it apparently goes into deep sleep mode but continues to power the peripherals then when I uncomment esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF); The micro is restarted and the program runs again from the beginning without waiting for the sleep time. What do I need to configure for it to work properly?

lewisxhe, 2025-01-24:
Paste your test code

DanielPerezGuzman22, 2025-01-24:
//Sleep

#include "LoRaBoards.h"

/*
Simple Deep Sleep with Timer Wake Up
=====================================
ESP32 offers a deep sleep mode for effective power
saving as power is an important factor for IoT
applications. In this mode CPUs, most of the RAM,
and all the digital peripherals which are clocked
from APB_CLK are powered off. The only parts of
the chip which can still be powered on are:
RTC controller, RTC peripherals ,and RTC memories

This code displays the most basic deep sleep with
a timer to wake it up and how to store data in
RTC memory to use it over reboots

This code is under Public Domain License.

Author:
Pranav Cherukupalli ***@***.***>
*/

#define uS_TO_S_FACTOR 1000000ULL /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  50         /* Time ESP32 will go to sleep (in seconds) */

RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason() {
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0:     Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1:     Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER:    Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD: Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP:      Serial.println("Wakeup caused by ULP program"); break;
    default:                        Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason); break;
  }

   // Actualizar la pantalla SH1106
  if (u8g2) {
    u8g2->clearBuffer(); // Limpia el buffer de la pantalla
    u8g2->setFont(u8g2_font_NokiaLargeBold_tf); // Selecciona la fuente
    u8g2->setCursor(0, 16);
    u8g2->print("Wakeup_reason: ");
    u8g2->sendBuffer(); // Envía los datos a la pantalla
    delay(3000);  //Take some time to open up the Serial Monitor
  }
}

void setup() {
  Serial.begin(115200);
  delay(1000);  //Take some time to open up the Serial Monitor

  setupBoards();                     // Inicializa la placa y la pantalla SH1106

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  /*
  First we configure the wake up source
  We set our ESP32 to wake up every 5 seconds
  */
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) + " Seconds");

    if (u8g2) {
    u8g2->clearBuffer(); // Limpia el buffer de la pantalla
    u8g2->setFont(u8g2_font_NokiaLargeBold_tf); // Selecciona la fuente
    u8g2->setCursor(0, 16);
    u8g2->print("Setup ESP32 to sleep for every ");
    u8g2->print(TIME_TO_SLEEP, 1);
    u8g2->print(" Seconds ");
    u8g2->sendBuffer(); // Envía los datos a la pantalla
    delay(3000);  //Take some time to open up the Serial Monitor
  }

  Serial.println("Going to sleep now");

      if (u8g2) {
    u8g2->clearBuffer(); // Limpia el buffer de la pantalla
    u8g2->setFont(u8g2_font_NokiaLargeBold_tf); // Selecciona la fuente
    u8g2->setCursor(0, 16);
    u8g2->print("Going to sleep now");
    u8g2->sendBuffer(); // Envía los datos a la pantalla
    delay(3000);  //Take some time to open up the Serial Monitor
  }

  /*
  Next we decide what all peripherals to shut down/keep on
  By default, ESP32 will automatically power down the peripherals
  not needed by the wakeup source, but if you want to be a poweruser
  this is for you. Read in detail at the API docs
  http://esp-idf.readthedocs.io/en/latest/api-reference/system/deep_sleep.html
  Left the line commented as an example of how to configure peripherals.
  The line below turns off all RTC peripherals in deep sleep.
  */
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  Serial.println("Configured all RTC Peripherals to be powered down in sleep");

  /*
  Now that we have setup a wake cause and if needed setup the
  peripherals state in deep sleep, we can now start going to
  deep sleep.
  In the case that no wake up sources were provided but deep
  sleep was started, it will sleep forever unless hardware
  reset occurs.
  */

  Serial.flush();
  if (u8g2) {
  u8g2->setPowerSave(1);
}

  delay(1000);
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop() {
  //This is not going to be called
}

lewisxhe, 2025-01-25:
esp_sleep_pd_config（ESP_PD_DOMAIN_RTC_PERIPH， ESP_PD_OPTION_OFF） will turn off the internal timer. At this time, the timed wake-up is invalid. Please see the document

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/low-power-mode/low-power-mode-soc.html#deep-sleep-mode

lewisxhe, 2025-01-25:
Your question is mentioned in the document. Now is the Chinese New Year. I will be on vacation during this period and will not reply to your question for the time being. Please refer to the document for solution.

DanielPerezGuzman22, 2025-01-25:
Thank you very much for your comments, I will review the documentation and try to solve the problems. We keep in touch. Greetings

closed 2025-03-11 by github-actions[bot]

DanielPerezGuzman22, 2025-03-21:
Dear all, I'm trying to make a weather station with the T Beam Supreme S3 but when I try to use the Deep Sleep code it connects on the first iteration and sends data correctly, but when the timeout elapses and it wakes up again it is unable to reconnect to the Gateway. In my case I'm not using TTN I use my own Gateway a RAK7289, in which I don't know how to reset "DevNonces" and what seems is that it doesn't store anything in "LWsession" and that's why it doesn't connect. Could you help me find the error, Here I share the code, thank you very much.

[RadioLib OTA.txt](https://github.com/user-attachments/files/19390536/RadioLib.OTA.txt)