Xinyuan-LilyGO/LilyGo-LoRa-Series #222: T-Beam v1.2: Cannot change baud rate on Neo-8 GNSS

moshe-braner, 2025-02-24:
A friend has a T-Beam v1.2 with the Ublox 8 GNSS module, that came with Meshtastic installed.  He flashed other firmware on it (SoftRF) that tries to communicate with the GNSS at 9600 baud, which is the Ublox factory default.  After much work, he found that:
 
* The Ublox is working, and communicating at 115200 baud
 
* Restoring the default settings with this tool:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/GPS/UBlox_Recovery
or similar code (changed to clear EEPROM not just flash) does not work:
 
* After restart the Ublox is back at 115200 baud.

So does this Neo-M8N module have the ability to save the settings (in non-volatile storage) for later restarts?  Is it possible that the settings have been written into non-changable storage (ROM, e-fuse)?

lewisxhe, 2025-02-25:
It can be set and saved, but I didn't write these examples because it would cause many people to make the GPS inaccessible and try to make the illusion that the GPS is broken.

The Meshtastic firmware will change the GPS baud rate to 9600, but the ublox-m8 defaults to 115200.

If you want to change the baud rate, please check the Ublox manual.
https://content.u-blox.com/sites/default/files/products/documents/[u-blox8-M8_ReceiverDescrProtSpec_UBX-13003221.pdf](https://content.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_UBX-13003221.pdf)

moshe-braner, 2025-02-25:
Is the default baud rate different for different versions of the U8?  I tried on a T-Beam I have, with U8 and sx1262, and after sending that U8 the "clear" commands, and restarting it, it works at 9600 baud.  But not so for my friend who has another similar T-Beam.

Also, that Ublox document says (on page 466) that the default baud rate is 9600.  Even if that is no longer true, maybe it was true for earlier versions of the U8?

lewisxhe, 2025-02-26:
My reply above is wrong. I consulted the supplier and the default baud rate is 9600. If it is changed to 115200, it should be changed by other programs.

moshe-braner, 2025-03-07:
Now reported by a friend:

"Today a new TBeam_V1.2 M8N  already flashed with SoftRF 1.3 arrived.
SoftRF did not 'see' the GPS.  On installing MB156 it worked."

- MB156 is my version of SoftRF, which now automatically searches for the baud rate used by the GNSS module.
-

lewisxhe, 2025-03-08:
SoftRF has been updated to v1.6, which may have fixed this problem. I will update the subsequent factory firmware to the latest version. Thank you for your report.

moshe-braner, 2025-03-08:
Looking at the source code for the "official" SoftRF version 1.6, I don't see any auto-baud-rate detection.  This has only been added in my fork.

closed 2025-04-21 by github-actions[bot]