Xinyuan-LilyGO/LilyGo-LoRa-Series #223: T3 S3 LR1121 - Lora

Doucie33, 2025-03-01:
J'ai acheter 5 modules T3 S3 LR1121, télécharger toutes les bibliothèques  sur IDE ARDUINO, Chercher tous les bon GPIO pour le module LORA, et impossible de faire fonctionner un simple envoie de message via la communication Lora , voici le code teste que j'utilise  :    

[image]

#include <Wire.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <RadioLib.h>

// 📌 Définition de l'écran OLED
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
#define OLED_RESET    -1
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

// 📌 Définition des broches I2C pour l'OLED et l'AXP2101
#define I2C_SDA 18
#define I2C_SCL 17
#define AXP2101_ADDRESS 0x3C  // Adresse confirmée de l'AXP2101

// 📡 Définition des broches LoRa LR1121
#define LORA_MOSI  6
#define LORA_MISO  3
#define LORA_SCK   5
#define LORA_NSS   7
#define LORA_RST   8
#define LORA_DIO1  33

// 📡 Fréquence LoRa
const float LORA_FREQUENCY = 915.0;

// 📡 Initialisation du module LoRa LR1121
SX1262 lora = new Module(LORA_NSS, LORA_DIO1, LORA_RST, LORA_BUSY);

// 🛠 Initialisation du matériel
void setup() {
    Wire.begin(I2C_SDA, I2C_SCL);
    SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_NSS);
    Serial.begin(115200);
    
    // 🔹 Démarrage OLED
    if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
        Serial.println("Erreur OLED !");
        return;
    }
    display.clearDisplay();
    display.setTextSize(1);
    display.setTextColor(WHITE);
    display.setCursor(0, 0);
    display.println("OLED OK !");
    display.display();
    delay(1000);

    // 🔹 Vérification AXP2101
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Test AXP2101...");
    display.display();
    delay(500);

    Wire.beginTransmission(AXP2101_ADDRESS);
    if (Wire.endTransmission() == 0) {
        display.println("AXP2101 OK !");
    } else {
        display.println("Erreur AXP2101 !");
        display.display();
        return;
    }
    display.display();
    delay(1000);

    // 🔹 Activation LoRa via AXP2101
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Activation LoRa...");
    display.display();
    delay(500);

    Wire.beginTransmission(AXP2101_ADDRESS);
    Wire.write(0x12); // Adresse du registre de contrôle
    Wire.write(0x04); // Activer l'alimentation du LoRa (ALDO2)
    if (Wire.endTransmission() == 0) {
        display.println("LoRa ON !");
    } else {
        display.println("Erreur LoRa ON !");
        display.display();
        return;
    }
    display.display();
    delay(2000);  // IMPORTANT : Attente pour s'assurer que le LoRa est bien activé

    // 🔹 Initialisation du module LoRa LR1121
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Init LoRa...");
    display.display();
    delay(500);

    int state = lora.begin(LORA_FREQUENCY, 125.0, 7, 5, 14, 8, 1.0, true);

    display.print("Code: ");
    display.println(state);
    if (state == RADIOLIB_ERR_NONE) {
        display.println("LoRa OK !");
    } else {
        display.println("Erreur LoRa !");
        display.display();
        return;
    }
    display.display();
    delay(2000);
}

// 🔁 Boucle principale
void loop() {
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Test LoRa envoi...");
    display.display();

    int state = lora.transmit("Hello LoRa!");

    if (state == RADIOLIB_ERR_NONE) {
        display.println("Message envoye !");
    } else {
        display.println("Echec envoi !");
        display.print("Code: ");
        display.println(state);
    }

    display.display();
    delay(5000);
}

lewisxhe, 2025-03-03:
Try changing DIO1 from 33 to 36
BUSY seems not defined by you, BUSY is 34

Doucie33, 2025-03-03:
J'ai remplacer le DIO1 pour 36 et défini BUSY en 34 , voici ce que j'obtiens .

[image]

[image]

[image]

[image]

[image]

[image]

et mon moniteur n'affiche rien : 

[image]

Voici le code : 

#include <Wire.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <RadioLib.h>

// 📌 Définition de l'écran OLED
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
#define OLED_RESET    -1
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

// 📌 Définition des broches I2C pour l'OLED et l'AXP2101
#define I2C_SDA 18
#define I2C_SCL 17
#define AXP2101_ADDRESS 0x3C  // Adresse confirmée de l'AXP2101

// 📡 Définition des broches LoRa LR1121
#define LORA_MOSI  6
#define LORA_MISO  3
#define LORA_SCK   5
#define LORA_NSS   7
#define LORA_RST   8
#define LORA_DIO1  36
#define LORA_BUSY  34

// 📡 Fréquence LoRa
const float LORA_FREQUENCY = 915.0;

// 📡 Initialisation du module LoRa LR1121
SX1262 lora = new Module(LORA_NSS, LORA_DIO1, LORA_RST, LORA_BUSY);

// 🛠 Initialisation du matériel
void setup() {
    Wire.begin(I2C_SDA, I2C_SCL);
    SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_NSS);
    Serial.begin(115200);
    
    // 🔹 Démarrage OLED
    if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
        Serial.println("Erreur OLED !");
        return;
    }
    display.clearDisplay();
    display.setTextSize(1);
    display.setTextColor(WHITE);
    display.setCursor(0, 0);
    display.println("OLED OK !");
    display.display();
    delay(1000);

    // 🔹 Vérification AXP2101
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Test AXP2101...");
    display.display();
    delay(500);

    Wire.beginTransmission(AXP2101_ADDRESS);
    if (Wire.endTransmission() == 0) {
        display.println("AXP2101 OK !");
    } else {
        display.println("Erreur AXP2101 !");
        display.display();
        return;
    }
    display.display();
    delay(1000);

    // 🔹 Activation LoRa via AXP2101
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Activation LoRa...");
    display.display();
    delay(500);

    Wire.beginTransmission(AXP2101_ADDRESS);
    Wire.write(0x12); // Adresse du registre de contrôle
    Wire.write(0x04); // Activer l'alimentation du LoRa (ALDO2)
    if (Wire.endTransmission() == 0) {
        display.println("LoRa ON !");
    } else {
        display.println("Erreur LoRa ON !");
        display.display();
        return;
    }
    display.display();
    delay(2000);  // IMPORTANT : Attente pour s'assurer que le LoRa est bien activé

    // 🔹 Initialisation du module LoRa LR1121
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Init LoRa...");
    display.display();
    delay(500);

    int state = lora.begin(LORA_FREQUENCY, 125.0, 7, 5, 14, 8, 1.0, true);

    display.print("Code: ");
    display.println(state);
    if (state == RADIOLIB_ERR_NONE) {
        display.println("LoRa OK !");
    } else {
        display.println("Erreur LoRa !");
        display.display();
        return;
    }
    display.display();
    delay(2000);
}

// 🔁 Boucle principale
void loop() {
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Test LoRa envoi...");
    display.display();

    int state = lora.transmit("Hello LoRa!");

    if (state == RADIOLIB_ERR_NONE) {
        display.println("Message envoye !");
    } else {
        display.println("Echec envoi !");
        display.print("Code: ");
        display.println(state);
    }

    display.display();
    delay(5000);
}

lewisxhe, 2025-03-03:
This one doesn't have axp, for t3 examples see here
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/T3S3Factory

Doucie33, 2025-03-03:
Quel est la procédure? Quel code dois-je télécharger ? faut-il télécharger tous les codes? dois-je utiliser le fichier  . ino?

EXIST-IL UN CODE PLUS SIMPLE POIR MON MODULE ? 

Je suis novice et soirée d'apprentissage.

lewisxhe, 2025-03-03:
Are you sure you bought LR1121?? Then why did you specify sx1262 when initializing?

// 📡 Initialisation du module LoRa LR1121
SX1262 lora = new Module(LORA_NSS, LORA_DIO1, LORA_RST, LORA_BUSY);

lewisxhe, 2025-03-03:
As a novice, shouldn't you start with the examples provided?
[image]

Doucie33, 2025-03-03:
[image]

voici ma commande

Doucie33, 2025-03-03:
j'ai essayé les exemples fournis et aucune confirmation possible, j'ai sélectionné dans utilitiés.h le bon module 
[image]

Doucie33, 2025-03-03:
j'ai corriger le code : 

#include <Wire.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <RadioLib.h>

// 📌 Définition de l'écran OLED
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
#define OLED_RESET    -1
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

// 📌 Définition des broches I2C pour l'OLED et l'AXP2101
#define I2C_SDA 18
#define I2C_SCL 17
#define AXP2101_ADDRESS 0x3C  // Adresse confirmée de l'AXP2101

// 📡 Définition des broches LoRa LR1121
#define LORA_MOSI  6
#define LORA_MISO  3
#define LORA_SCK   5
#define LORA_NSS   7
#define LORA_RST   8
#define LORA_DIO1  36
#define LORA_BUSY  34

// 📡 Fréquence LoRa
const float LORA_FREQUENCY = 915.0;

// 📡 Initialisation du module LoRa LR1121
LR1121 lora = new Module(LORA_NSS, LORA_DIO1, LORA_RST, LORA_BUSY);

// 🛠 Initialisation du matériel
void setup() {
    Wire.begin(I2C_SDA, I2C_SCL);
    SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_NSS);
    Serial.begin(115200);
    
    // 🔹 Démarrage OLED
    if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
        Serial.println("Erreur OLED !");
        return;
    }
    display.clearDisplay();
    display.setTextSize(1);
    display.setTextColor(WHITE);
    display.setCursor(0, 0);
    display.println("OLED OK !");
    display.display();
    delay(1000);

    // 🔹 Vérification AXP2101
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Test AXP2101...");
    display.display();
    delay(500);

    Wire.beginTransmission(AXP2101_ADDRESS);
    if (Wire.endTransmission() == 0) {
        display.println("AXP2101 OK !");
    } else {
        display.println("Erreur AXP2101 !");
        display.display();
        return;
    }
    display.display();
    delay(1000);

    // 🔹 Activation LoRa via AXP2101
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Activation LoRa...");
    display.display();
    delay(500);

    Wire.beginTransmission(AXP2101_ADDRESS);
    Wire.write(0x12); // Adresse du registre de contrôle
    Wire.write(0x04); // Activer l'alimentation du LoRa (ALDO2)
    if (Wire.endTransmission() == 0) {
        display.println("LoRa ON !");
    } else {
        display.println("Erreur LoRa ON !");
        display.display();
        return;
    }
    display.display();
    delay(2000);  // IMPORTANT : Attente pour s'assurer que le LoRa est bien activé

    // 🔹 Initialisation du module LoRa LR1121
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Init LoRa...");
    display.display();
    delay(500);

    int state = lora.begin(LORA_FREQUENCY, 125.0, 7, 5, 14, 8, 1.0, true);

    display.print("Code: ");
    display.println(state);
    if (state == RADIOLIB_ERR_NONE) {
        display.println("LoRa OK !");
    } else {
        display.println("Erreur LoRa !");
        display.display();
        return;
    }
    display.display();
    delay(2000);
}

// 🔁 Boucle principale
void loop() {
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Test LoRa envoi...");
    display.display();

    int state = lora.transmit("Hello LoRa!");

    if (state == RADIOLIB_ERR_NONE) {
        display.println("Message envoye !");
    } else {
        display.println("Echec envoi !");
        display.print("Code: ");
        display.println(state);
    }

    display.display();
    delay(5000);
}

[image]

lewisxhe, 2025-03-03:
```

#include <Wire.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <RadioLib.h>

// 📌 Définition de l'écran OLED
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
#define OLED_RESET -1
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

// 📌 Définition des broches I2C pour l'OLED et l'AXP2101
#define I2C_SDA 18
#define I2C_SCL 17

// 📡 Définition des broches LoRa LR1121
#define LORA_MOSI 6
#define LORA_MISO 3
#define LORA_SCK 5
#define LORA_NSS 7
#define LORA_RST 8
#define LORA_DIO1 36
#define LORA_BUSY 34

// 📡 Fréquence LoRa
const float LORA_FREQUENCY = 915.0;

// 📡 Initialisation du module LoRa LR1121
LR1121 lora = new Module(LORA_NSS, LORA_DIO1, LORA_RST, LORA_BUSY);

bool            transmittedFlag = false;
void setFlag(void)
{
    // we got a packet, set the flag
    transmittedFlag = true;
}

// 🛠 Initialisation du matériel
void setup()
{
    Wire.begin(I2C_SDA, I2C_SCL);
    SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI);
    Serial.begin(115200);

    // 🔹 Démarrage OLED
    if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
        Serial.println("Erreur OLED !");
        return;
    }
    display.clearDisplay();
    display.setTextSize(1);
    display.setTextColor(WHITE);
    display.setCursor(0, 0);
    display.println("OLED OK !");
    display.display();
    delay(1000);

    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Activation LoRa...");
    display.display();
    delay(500);

    // 🔹 Initialisation du module LoRa LR1121
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Init LoRa...");
    display.display();
    delay(500);

    int state = lora.begin(LORA_FREQUENCY, 125.0, 7, 5, 14, 8);

    display.print("Code: ");
    display.println(state);
    if (state == RADIOLIB_ERR_NONE) {
        display.println("LoRa OK !");
    } else {
        display.println("Erreur LoRa !");
        display.display();
        return;
    }
    Serial.println("LoRa OK!@");
    display.display();
    delay(2000);

    // LR1121
    // set RF switch configuration for Wio WM1110
    // Wio WM1110 uses DIO5 and DIO6 for RF switching
    static const uint32_t rfswitch_dio_pins[] = {
        RADIOLIB_LR11X0_DIO5, RADIOLIB_LR11X0_DIO6,
        RADIOLIB_NC, RADIOLIB_NC, RADIOLIB_NC
    };

    static const Module::RfSwitchMode_t rfswitch_table[] = {
        // mode                  DIO5  DIO6
        { LR11x0::MODE_STBY,   { LOW,  LOW  } },
        { LR11x0::MODE_RX,     { HIGH, LOW  } },
        { LR11x0::MODE_TX,     { LOW,  HIGH } },
        { LR11x0::MODE_TX_HP,  { LOW,  HIGH } },
        { LR11x0::MODE_TX_HF,  { LOW,  LOW  } },
        { LR11x0::MODE_GNSS,   { LOW,  LOW  } },
        { LR11x0::MODE_WIFI,   { LOW,  LOW  } },
        END_OF_MODE_TABLE,
    };
    lora.setRfSwitchTable(rfswitch_dio_pins, rfswitch_table);

    // LR1121 TCXO Voltage 2.85~3.15V
    lora.setTCXO(3.0);

    lora.setPacketSentAction(setFlag);

    lora.transmit("Hello LoRa!");
    Serial.println("Start...");
}

// 🔁 Boucle principale
void loop()
{
    if (transmittedFlag) {
        // reset flag
        transmittedFlag = false;
        display.clearDisplay();
        display.setCursor(0, 0);
        display.println("Test LoRa envoi...");
        display.display();

        lora.finishTransmit();

        delay(1000);

        int state = lora.transmit("Hello LoRa!");

        if (state == RADIOLIB_ERR_NONE) {
            display.println("Message envoye !");
            Serial.println("Send OK!");
        } else {
            display.println("Echec envoi !");
            display.print("Code: ");
            display.println(state);
            Serial.printf("LoRa failed %d\n", state);
        }

        display.display();
    }
}

```

Doucie33, 2025-03-05:
Salut, 

Merci pour ton aide très précieuse ton code fonctionne et j'ai pu tester mes T3S3 : 

 
[image]

j'aurais aimer des explication sur le code pour comprendre mon erreur afin de m'améliorer dans ma démarche.
J'ai également apporter des modification dans le code et je cherche a améliorer la porter de communication car elle est très courte (10 maitres +-).

 #include <Wire.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <RadioLib.h>
#include <DHT.h>

// 📌 Définition de l'écran OLED
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64
#define OLED_RESET -1
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

// 📌 Définition des broches I2C pour l'OLED et l'AXP2101
#define I2C_SDA 18
#define I2C_SCL 17

// 📡 Définition des broches LoRa LR1121
#define LORA_MOSI 6
#define LORA_MISO 3
#define LORA_SCK 5
#define LORA_NSS 7
#define LORA_RST 8
#define LORA_DIO1 36
#define LORA_BUSY 34

// 📡 Fréquence LoRa
const float LORA_FREQUENCY = 915.0;

// 📡 Initialisation du module LoRa LR1121
LR1121 lora = new Module(LORA_NSS, LORA_DIO1, LORA_RST, LORA_BUSY);

// 🌡️ Définition du capteur DHT22
#define DHTPIN 47      // ✅ Nouveau GPIO pour le DHT22
#define DHTTYPE DHT22  // 🔹 Type de capteur DHT
DHT dht(DHTPIN, DHTTYPE);

bool transmittedFlag = false;
int messageCount = 0; // 🔹 Compteur de messages envoyés

void setFlag(void)
{
    transmittedFlag = true; // Un paquet a été envoyé
}

// 🛠 Initialisation du matériel
void setup()
{
    Wire.begin(I2C_SDA, I2C_SCL);
    SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI);
    Serial.begin(115200);

    // 🔹 Démarrage OLED
    if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
        Serial.println("Erreur OLED !");
        return;
    }
    display.clearDisplay();
    display.setTextSize(1);
    display.setTextColor(WHITE);
    display.setCursor(0, 0);
    display.println("OLED OK !");
    display.display();
    delay(1000);

    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Activation LoRa...");
    display.display();
    delay(500);

    // 🔹 Initialisation du module LoRa LR1121
    display.clearDisplay();
    display.setCursor(0, 0);
    display.println("Init LoRa...");
    display.display();
    delay(500);

    int state = lora.begin(LORA_FREQUENCY, 125.0, 7, 5, 14, 8);

    display.print("Code: ");
    display.println(state);
    if (state == RADIOLIB_ERR_NONE) {
        display.println("LoRa OK !");
    } else {
        display.println("Erreur LoRa !");
        display.display();
        return;
    }
    Serial.println("LoRa OK!");
    display.display();
    delay(2000);

    // Configuration du RF Switch pour LR1121
    static const uint32_t rfswitch_dio_pins[] = {
        RADIOLIB_LR11X0_DIO5, RADIOLIB_LR11X0_DIO6,
        RADIOLIB_NC, RADIOLIB_NC, RADIOLIB_NC
    };

    static const Module::RfSwitchMode_t rfswitch_table[] = {
        { LR11x0::MODE_STBY,   { LOW,  LOW  } },
        { LR11x0::MODE_RX,     { HIGH, LOW  } },
        { LR11x0::MODE_TX,     { LOW,  HIGH } },
        { LR11x0::MODE_TX_HP,  { LOW,  HIGH } },
        { LR11x0::MODE_TX_HF,  { LOW,  LOW  } },
        { LR11x0::MODE_GNSS,   { LOW,  LOW  } },
        { LR11x0::MODE_WIFI,   { LOW,  LOW  } },
        END_OF_MODE_TABLE,
    };
    lora.setRfSwitchTable(rfswitch_dio_pins, rfswitch_table);

    // LR1121 TCXO Voltage 2.85~3.15V
    lora.setTCXO(3.0);

    lora.setPacketSentAction(setFlag);

    // 🌡️ Démarrage du capteur DHT22
    dht.begin();

    // Premier envoi
    messageCount++; // 🔹 Incrémente dès le premier message
}

// 🔁 Boucle principale
void loop()
{
    if (transmittedFlag) {
        transmittedFlag = false;
        messageCount++; // 🔹 Incrémentation du compteur

        // 🌡️ Lecture des données du DHT22
        float temperature = dht.readTemperature();
        float humidity = dht.readHumidity();

        // Vérification des erreurs de lecture
        if (isnan(temperature) || isnan(humidity)) {
            Serial.println("❌ Erreur de lecture DHT22 !");
            display.clearDisplay();
            display.setCursor(0, 0);
            display.println("❌ DHT22 Erreur !");
            display.display();
            delay(5000);
            return;
        }

        // 🔹 Création du message avec les données du DHT22
        String message = "Hello LoRa! mess : " + String(messageCount) +
                         ", Temp: " + String(temperature, 1) + "°C" +
                         ", Hum: " + String(humidity, 1) + "%";

        Serial.println("📡 Envoi du message :");
        Serial.println(message);

        display.clearDisplay();
        display.setCursor(0, 0);
        display.print("Message #: ");
        display.println(messageCount);
        display.print("Temp: ");
        display.print(temperature, 1);
        display.println(" C");
        display.print("Hum: ");
        display.print(humidity, 1);
        display.println(" %");
        display.display();

        int state = lora.transmit(message);

        if (state == RADIOLIB_ERR_NONE) {
            display.println("✅ Envoye !");
            Serial.println("Send OK!");
        } else {
            display.println("❌ Echec !");
            Serial.printf("LoRa failed %d\n", state);
        }

        display.display();
        delay(5000); // 🔹 Attente de 5 secondes avant le prochain envoi
    }
}

Merci encore une foi de ton aide.

lewisxhe, 2025-03-05:
I still say the same thing as before, please see T3S3Factory.ino, lora parameters affect the distance and transmission speed. Please set reasonable power and other parameters according to the laws of your country

[image]

netmonk, 2025-03-08:
Standard antenna shipped with those devices are often crap. Cheap means no QA at the end of production line. 

So basically, 10m i would consider that one of the antenna is not working at all.

Doucie33, 2025-03-08:
Ça veux dire que je dois me faire rembourser tous mes modules ?

lewisxhe, 2025-03-10:
What is less than 10 meters? Have you checked your code? The transmission power you set is much lower than the default transmission power.

And I also said that the default factory communication distance is at least 1KM. Of course, this is the distance in the open air.

[image]

Doucie33, 2025-03-12:
Salut, 

 float freq = LORA_FREQUENCY;    devrait-il pas êtres la fréquence du pays ? soit 915 pour le Québec ?

[image]

J'avais essais ce code et pas de changement sur la distance

lewisxhe, 2025-03-13:
Did you connect the antenna to the SMA interface? Even the worst antenna can't have a range of only 10 meters.

lyusupov, 2025-03-27:
@Doucie33 

make sure that you supply `true` as a `forceHighPower` argument for _setOutputPower(int8_t power, bool forceHighPower)_ method

https://github.com/jgromes/RadioLib/pull/1409

[image]

closed 2025-05-11 by github-actions[bot]