Xinyuan-LilyGO/LilyGo-LoRa-Series #233: Trying to build a template for tinygs to get T-Beam Supreme L76K 433MHz to work with it

stephenmorton, 2025-04-18:
Hi,
I am having a lot of trouble building a template for TinyGS with the Lilygo T-Beam supreme L76K 443MHz.
I have an older T-Beam and it is working fine on tinyGS, but it seems tinyGS is not auto discovering this board and not auto configuring itself.

The instructions for building a Tempate are here https://github.com/G4lile0/tinyGS/wiki/Board-Templates

An example template for a TTGO LoRa 32 V2 is:-

{"name":"[433] TTGO LoRA 32 v2","aADDR":60,"oSDA":21,"oSCL":22,"oRST":16,"pBut":0,"led":22,"radio":1,"lNSS":18,"lDIO0":26,"lDIO1":33,"lBUSSY":0,"lRST":14,"lMISO":19,"lMOSI":27,"lSCK":5,"lTCXOV":0.0}

The template I have built, from looking at your schematic diagrams, product page and googling is:-

{"name":"[433] TTGO T-Beam Sup","aADDR":60,"oSDA":17,"oSCL":18,"oRST":16,"pBut":0,"led":22,"radio":0,"lNSS":18,"lDIO0":26,"lDIO1":33,"lBUSSY":0,"lRST":14,"lMISO":19,"lMOSI":27,"lSCK":5,"lTCXOV":5.0}

I think my template is close, but looking at your schematic and info, I could not find the following bits of information to complete the template and I think I have guesses wrong for the following items (the other bits are fine, things like name is fine)

Would you pls be able to help me determine the correct values for the following?
aADDR: I2C address of the OLED display (in decimal format)    *** I think 60
oSDA: OLED SDA pin    *** I think 17
oSCL: OLED SCL pin    *** I think 18
oRST: OLED RST pin    *** I think 18
pBut: GPIO used for the board user button    *** I think 0
led: GPIO used for the main board indicator led    *** I think 22
lNSS: LoRa NSS pin    *** I think 18
lDIO0: LoRa DIO0 pin    *** I think 26
lDIO1: LoRa DIO1 pin    *** I think 33
lBUSSY: LoRa BUSY pin    *** I think 0
lRST: LoRa RST pin    *** I think 14
lMISO: LoRa MISO pin    *** I think 19
lMOSI: LoRa MOSI pin    *** I think 27
lSCK: LoRa SCK pin    *** I think 5
lTCXOV: LoRa TXCO voltage (float value, only used for sx126x modules)    *** I think after googling it, it should be 5.0
Any help on the above you could provide would be super appreciated and would allow me to publish a template so outhers with this board can use it with TinyGS.
Thanks,

Steve

lewisxhe, 2025-04-19:
The power channels of T-Beam supreme and
T-Beam are different. If you don't turn on the power, it can't be used. Please refer to the code to see which channel the power uses

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/c47ef9d5d34aee08b7cd96de19ed676c3fce6b59/examples/RadioLibExamples/Receive_Interrupt/LoRaBoards.cpp#L170

For the GPIO list, refer to here

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/c47ef9d5d34aee08b7cd96de19ed676c3fce6b59/examples/RadioLibExamples/Receive_Interrupt/utilities.h#L392

stephenmorton, 2025-04-21:
Thanks for the info Lewis, that helps a lot.

Steve

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/c47ef9d5d34aee08b7cd96de19ed676c3fce6b59/examples/RadioLibExamples/Receive_Interrupt/LoRaBoards.cpp#L170

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/c47ef9d5d34aee08b7cd96de19ed676c3fce6b59/examples/RadioLibExamples/Receive_Interrupt/utilities.h#L392

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/c47ef9d5d34aee08b7cd96de19ed676c3fce6b59/examples/RadioLibExamples/Receive_Interrupt/LoRaBoards.cpp#L170

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/c47ef9d5d34aee08b7cd96de19ed676c3fce6b59/examples/RadioLibExamples/Receive_Interrupt/utilities.h#L392