Xinyuan-LilyGO/LilyGo-LoRa-Series #234: Not Recognising Battery Installed and Not Holding Charge

FPVKernow, 2025-04-24:
Hello,

I have been having some battery issues with my LILYGO SoftRF T-Beam SUPREME ESP32-S3.

My device doesn’t seem to be recognising it has a battery and won’t hold its change if it does.

I’ve tried putting the latest firmware on and have fully charged the battery and has been checked with a multimeter.

I have emailed the person running the GitHub for SoftRF FLARM (https://github.com/lyusupov/SoftRF) which the T-Beam Supreme is used for but he's not able to help at this time.

I can upload a video of the issue if that helps.

I really appreciate any advice at this time.

Many thanks,

Tom

lewisxhe, 2025-04-25:
I suspect the MOSFET is damaged. Are you willing to give it a try?

Short-circuit the two pins of the MOS as shown below. The battery should be detected and the power supply should be normal.

[image]

FPVKernow, 2025-04-25:
Hi Lewisxhe,

I tried shorting the MOSFET and it was tricky to get the connection. The first few attempts it said Battery N/A. I was able to get it to say Battery 2.4 but then it shut down quickly afterwards. I have checked the 18650 battery with a multimeter and it says 3.3V.

The device does not boot from the battery while the MOS is short-circuit.

Kind regards,
Tom

FPVKernow, 2025-04-25:
I see 2.45 volts.

lewisxhe, 2025-04-26:
What voltage is the battery measuring?

FPVKernow, 2025-04-29:
Battery is measuring 3.3V when I take it out of the device.

lewisxhe, 2025-04-30:
OK. That should be a problem with the MOSFET. You can directly remove the MOSFET in the picture and short-circuit the two points. Then measure the voltage of the core board again, it should be 3.3V.
By the way, 3.3V is a bit low.

The function of the MOSFET is to prevent the battery from being reversed. When removing the MOSFET, you need to pay attention to the polarity when installing the battery.

FPVKernow, 2025-05-13:
Hello, sorry for the delay getting back to you.

I have removed the MOSFET and short-circuit the two pins. The device still does not boot on battery alone and when powered over USB, it is saying BAT N/A. Checking the voltage on the battery after charging with a separate charger it is 3.3V.

At this point, I feel replacing the hardware will be the easiest option.

Thank you for your help.

lewisxhe, 2025-05-14:
Well, there is no other way, contact the dealer for replacement.

closed 2025-05-14 by lewisxhe