Xinyuan-LilyGO/LilyGo-LoRa-Series #235: Antenna Connection Clarification for T-T3S3 LR1121 (Sub-GHz SMA/IPEX & 2.4 GHz)

patricklouvel, 2025-04-28:
Hello LilyGo Team,

I have some questions regarding the intended antenna connections for the LilyGo T-T3S3 LR1121 board, especially concerning its dual-band LoRa capabilities (sub-GHz and 2.4 GHz).

Looking at the board and the schematic, I observe the following:

There is one SMA connector and one IPEX (u.FL) connector apparently dedicated to LoRa.

On my specific board (T3S3 V1.2), a 0-ohm resistor seems to connect the RF path only to the IPEX connector, leaving the SMA path unpopulated.

The Semtech LR1121 chip itself is capable of both sub-GHz and 2.4 GHz operation. Datasheets for the LR1121 sometimes suggest a separate RF path or switch for 2.4 GHz, possibly using an IPEX4 connector directly on the module.

My questions are:

- [ ] **Sub-GHz Antenna (e.g., 868/915 MHz):**

Can either the SMA or the IPEX connector be used for the sub-GHz LoRa antenna?

Does the selection depend solely on the placement of the 0-ohm resistor (meaning only one connector is active for sub-GHz at any given time), or is there another selection mechanism?

- [ ] **2.4 GHz Antenna:**

Where is the antenna for 2.4 GHz LoRa operation intended to be connected?

Is it meant to be attached to an IPEX4 connector directly on the LR1121 module used on the T3S3 board (if that connector is present/accessible)? Or does it share the path with the sub-GHz connectors through an internal RF switch?

- [ ] **Simultaneous Use:**

To fully utilize the LR1121's capabilities, can both a sub-GHz antenna (connected to the active SMA or IPEX port) and a 2.4 GHz antenna (connected to its designated port) be attached and functional at the same time? Or does the user need to physically swap antennas or components depending on the desired operating band?

Could you please provide clarification on the correct usage of the antenna ports for both sub-GHz and 2.4 GHz LoRa operation on the T-T3S3 LR1121?

Thank you for your support!

lyusupov, 2025-04-28:
[image]

- Sub-1GHz is exposed to **pad on the module**. Thus it is available on T3S3 PCB as well
- 2.x GHz is exposed to IPEX (uFL) connector **on the module only**

patricklouvel, 2025-04-28:
Thank you for the previous clarification regarding the antenna paths:

Sub-GHz is routed from the module pad to the main T3S3 PCB, where both SMA and IPEX (u.FL) connector footprints are available.

2.4 GHz is routed to a separate IPEX connector on the module itself.

This helps understand the routing. However, there is still ambiguity regarding the Sub-GHz connection on the main PCB:

Since only one path (either to SMA or to the PCB IPEX) can be active at a time for the Sub-GHz signal (typically selected by a 0-ohm resistor), could you please clarify:

Which connector (SMA or the main PCB IPEX) is the intended default active connection for Sub-GHz LoRa as the board ships from the factory?

Given that an SMA antenna is often supplied for Sub-GHz, many users naturally assume the SMA port is the default active connection. Does the placement of the 0-ohm resistor definitively indicate the active port, potentially overriding the assumption based on the included antenna? (e.g., If the resistor is visually confirmed to be linking the IPEX path, should users trust that visual confirmation and use the IPEX port, even if an SMA antenna was provided?)

This clarification would be very helpful for users to ensure they connect the Sub-GHz antenna to the correct, active port, preventing potential signal issues or damage from transmitting without a proper load.

Thank you!

lewisxhe, 2025-04-29:
By default, SMA is connected and IPEX is disconnected.