Xinyuan-LilyGO/LilyGo-LoRa-Series #236: T3 S3 board design issue

PovilasID, 2025-05-08:
Hey,

Pushing in JST battery connector is fine but there is obstruction if you try to pull it out.

[image]
If you try to pull out the connector it gets snagged on this piece and strips the wires out of the connector. 
To make more user friendly you could either relocate that elements or flip the connector to face the outside. All the other connectors are facing that way and it would void any obstructions.

lewisxhe, 2025-05-08:
This issue has been reported, and the current version will not be changed. The following version iterations will correct this issue

lewisxhe, 2025-05-09:
[image]
As an iterative version it would look like this

PovilasID, 2025-05-09:
Oh very cool.
Two questions:

[image]
1 looks like a battery connector but a more substantial one? Is it JST 2.0 mm?
2 Looks like a new connector? What kind of standard is it using? What is it for? Also power?

lewisxhe, 2025-05-09:
1. 2.00mm
2. Supports 4.4V~6V solar panel voltage input, changes the charging scheme. Can also be used as external power input

PovilasID, 2025-05-09:
@lewisxhe dam should have waited... can I trade in 1.2 version ? 😆

lewisxhe, 2025-05-09:
Probably not, or you can wait. Not sold yet

nirshem256, 2025-05-26:
Don't waste your time, Lilygo doesn't give software support for their products, you can buy here only non baked hardware...

closed 2025-06-03 by lewisxhe

mitra42, 2025-10-27:
Sorrey @PovilasID  - could you clarify, are you using the new TCXO or saying that you have a different experience using the V1.2 (which is other than the USB issue and the easily knocked off WiFi antenna, a very nice board)