Xinyuan-LilyGO/LilyGo-LoRa-Series #245: Footprint info for T3 S3 v1.2?

claudobahn, 2025-06-16:
Hi @lewisxhe is the footprint for T3S3 v1.2 the same as for https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/dimensions/T3_V1.62.dwg ? I'm trying to create a footprint in KiCad in order to make a PCB w/ headers for the T3S3 V1.2.

It does seem like you've changed the drill pattern for the T3S3 v1.3 board, at least if I'm reading the DXF you've posted for it correctly. So unfortunately I can't use that as a reference, I think.

As a side note, have you ever considered publishing KiCad symbols and footprints like Espressif does? Would make it easier to integrate your products, I think? https://github.com/espressif/kicad-libraries

lewisxhe, 2025-06-16:
V1.3 is identical to v1.2.

claudobahn, 2025-06-17:
Thanks. Do you have a DXF, DWG or even a STEP file for T3S3 v1.2? Only found the V1.3 and not the v1.2 in the repo.

lewisxhe, 2025-06-17:
v1.2 and v1.3 are the same