Xinyuan-LilyGO/LilyGo-LoRa-Series #247: What are the IO pin numbers for RXD and TXD?

ckdyne, 2025-07-01:
Hi there, simple question but I'm having trouble finding a definitive answer online. On the T-Beam Supreme there are two pins labeled RXD and TXD, on the left side of the OLED at the bottom of the row of pins ("left" meaning with respect to the pin labels being right-side up). The only schematic for the T-Beam Supreme that I can find labels all other RX and TX pins (GPS_RX_PIN and GPS_TX_PIN are IO09 and IO08 respectively, and on the bottom of the board there is a port with GND, DC1, TXD and RXD pins, being IO43 and IO44) save for these two. I want to use them for UART communication to an external serial device, so I need to specify these pins in my script. What are the pin numbers for these? Also to clarify, these can be used for UART, correct?

https://lilygo.cc/products/t-beam-supreme

Schematic I'm referring to, found on the page linked above:

[image]

lewisxhe, 2025-07-02:
[image]

lewisxhe, 2025-07-04:
Free mapping, any free GPIO can be set to RX, TX

moshe-braner, 2025-07-31:
I am not familiar with the T-Beam Supreme, but on the plain T-Beam you cannot use the "RX" pin for UART input from some other device, since it is connected internally to the USB port.  Those are normally handled as UART0.  And UART1 is normally used for the GNSS.  Thus if you need 2-way communications to some other device, use UART2, and you can set it up to use any available GPIO pins.

lewisxhe, 2025-08-02:
UART0 is idle. USBC is the hardware port of ESP32s3, not UART0.

closed 2025-09-17 by github-actions[bot]