Xinyuan-LilyGO/LilyGo-LoRa-Series #248: T-Beam Supreme PMU Access

ecniv, 2025-07-08:
Hi,
I have two T-Beam Supreme devices (one L76K and one ublox) and I'm trying to access the PMU for battery voltage monitoring in my firmware, but failing miserably over the last few days. Most recently I have been testing the L76K, and here are my results:

What works:
- Board powers on/off correctly with battery
- OLED display works
- I2C scan shows devices at 0x1C, 0x3C, 0x77
- When I load Meshtastic firmware it works great - it seems to have access to battery, the blue LED, charging, etc. all work. However my own firmware doesn't get any of that.

What doesn't work:
- Cannot initialize PMU using XPowersLib with either AXP2101 or AXP192 at any address
- Tried addresses 0x1C, 0x34, 0x35, 0x36, 0x37
- Both PMU classes fail to initialize, returning 0.00V and 0% battery
- Cannot control the blue LED (Meshtastic firmware can control it, but our Arduino code cannot)

Questions:
- What is the exact PMU chip model on T-Beam Supreme v1.0?
- What I2C address does the PMU use?
- Are there any enable pins or special initialization required?
- Can you provide working Arduino code for reading battery voltage?
- Which library should I use - XPowersLib or something else?
- Is there a specific version of XPowersLib that works with your hardware?
- How do I control the blue LED? What GPIO pin is it connected to?
- Are there any other hardware features that require special initialization?

My setup:
- T-Beam Supreme v1.0
- Arduino/PlatformIO
- XPowersLib v0.3.0
- Pins: PMU I2C on 17/18, OLED on 41/40

I2C Scan Results:
- 0x1C: Unknown device (possibly PMU?)
- 0x3C: OLED display (working)
- 0x77: Unknown device (sensor?)

The fact that the board powers correctly with battery but I can't access the PMU suggests there's either a different chip model or specific initialization required that's not documented. Similarly, the blue LED works with Meshtastic but not with our Arduino code.
Any help would be greatly appreciated!

Thanks!

lewisxhe, 2025-07-08:
Completely opposite. PMU SDA 42, SCL 41, see pin list
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/docs/en/t_beam_supreme/t_beam_supreme_hw.md

ecniv, 2025-07-08:
@lewisxhe thank you!! Somehow I did not see that documentation. I have sorted things out now. Thanks for the quick reply.