Xinyuan-LilyGO/LilyGo-LoRa-Series #258: T-Beam V1.2 does not transmit any LoRa packets (only receiving works)

Novusvetus, 2025-09-05:
# T-Beam V1.2 does not transmit any LoRa packets (only receiving works)

## Describe the bug
My LILYGO TTGO Meshtastic T-Beam V1.2 is able to receive LoRa packets but does not transmit anything. Other Meshtastic devices do not detect it, and a sniffer device I built shows no outgoing packets.

## To Reproduce
Steps I have taken:
1. Flashed the latest Meshtastic firmware (several times)
2. Also flashed the standard LILYGO firmware as suggested by support (https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/hardware_test/t-beam)
3. Performed multiple factory resets
4. Tested together with multiple working Meshtastic devices
5. Used a separate device to monitor LoRa traffic (no transmissions from this T-Beam)

## Expected behavior
The T-Beam should transmit LoRa packets and be visible to other Meshtastic devices.  

## Actual behavior
- Device receives data correctly  
- Device does not transmit any LoRa packets  
- Other devices cannot see it in the mesh network  

## Device details
- Model: LILYGO TTGO Meshtastic T-Beam V1.2  
- Firmware tested: latest Meshtastic release, latest standard firmware from LILYGO  
- Reset to factory settings: yes  

## Additional context
All my other Meshtastic devices work as expected with the same firmware versions.  
Only this unit shows the issue, which seems to indicate a hardware defect in the transmission path.  

I would like to try solving this here first, before I return the device as defective.  

## Logging of the testing firmware
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
setupBoards
-----------------------------------
Reset reason: In case of deep sleep, reset was not caused by exit from deep sleep
PSRAM is enable! PSRAM: 4.00MB
Flash:4 MB
Flash speed:40 M
Model:ESP32-D0WDQ6-V3
Chip Revision:3
Freq:240 MHZ
SDK Ver:v4.4.7-dirty
DATE:Dec 18 2024
TIME:16:08:12
EFUSE MAC: A0:DD:6C:XX:XX:XX
-----------------------------------
Scan Wire...
I2C Devices scanning
	Find AXP192/AXP2101 PMU!
	Find SSD1306/SH1106 dispaly!
Scan devices done.

AXP2101 PMU init succeeded, using AXP2101 PMU
=========================================
DC1  : +   Voltage: 3300 mV 
DC2  : -   Voltage: 0500 mV 
DC3  : -   Voltage: 0500 mV 
DC4  : -   Voltage: 1800 mV 
DC5  : -   Voltage: 3300 mV 
ALDO1: -   Voltage: 1800 mV 
ALDO2: +   Voltage: 3300 mV 
ALDO3: +   Voltage: 3300 mV 
ALDO4: -   Voltage: 3300 mV 
BLDO1: -   Voltage: 1800 mV 
BLDO2: -   Voltage: 3300 mV 
=========================================
PowerKeyPressOffTime:4 Second
Try to init L76K . Wait stop .
Try to init L76K . Wait stop ..
Try to init L76K . Wait stop ..
Update baudrate : 9600
Get ack successes!
Get ack successes!
Get ack successes!
Get ack successes!
UBlox GNSS init succeeded, using UBlox GNSS Module

init done . 
Starting BLE:T-Beam-A0:DD
Characteristic defined! Now you can read it in your phone!
[SX1276]: Initializing ... success!
[Radio] Starting to listen ... [  8932][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
[ 13314][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
[ 17696][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
[ 22078][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
prevButtonHandleEvent currentFrames:0 frames_count:6
currentFrames : 1
Start transmit
transmission finished!
[Radio] Sending another packet ... Radio TX done !
[ 26461][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
transmission finished!
[Radio] Sending another packet ... Radio TX done !
transmission finished!
[Radio] Sending another packet ... Radio TX done !
transmission finished!
[Radio] Sending another packet ... Radio TX done !
transmission finished!
[Radio] Sending another packet ... [ 30843][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
Radio TX done !
transmission finished!
[Radio] Sending another packet ... Radio TX done !
transmission finished!
[Radio] Sending another packet ... Radio TX done !
transmission finished!
[Radio] Sending another packet ... Radio TX done !
[ 35226][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
transmission finished!
[Radio] Sending another packet ... Radio TX done !
transmission finished!
[Radio] Sending another packet ... Radio TX done !
transmission finished!
```

lewisxhe, 2025-09-05:
The test firmware shows "Tx Done," indicating that the transmission has already occurred. Do you have other T-beams?

What are the RF module models for your other Meshtastic nodes?

Novusvetus, 2025-09-05:
Thank you for the response.

To answer your questions:
- I do not have other T-Beam devices. My other Meshtastic nodes are:
  - Seeed Studio Card Tracker T1000-E
  - RAK WisBlock devices
- In addition, I built a sniffer based on a Raspberry Pi with existing LoRa hardware. With this setup I can confirm that **no packets are being transmitted** from the T-Beam V1.2, even though the firmware reports *"Tx Done."*

The other devices communicate with each other without problems. Only the T-Beam is unable to transmit packets into the mesh.

I am open to further debugging steps or tests you would like me to run.

lewisxhe, 2025-09-06:
I checked the Seeed Studio Card Tracker T1000-E, which is based on the LR1110. If the radio parameters don't match, it won't be able to search for data packets. My intuition suggests this is a compatibility issue with the radio module. Do you have other boards based on the SX1278? The SX1278 is a first-generation LoRa modem chip, and there are some compatibility issues with the second-generation version.

closed 2025-10-22 by github-actions[bot]