Xinyuan-LilyGO/LilyGo-LoRa-Series #259: Connecting T Beam Supreme Lorawan without restarting DevNounces

DanielPerezGuzman22, 2025-09-05:
Dear friends, I'm trying to connect a T beam Supreme S3 to a Rak as a waterway via LORAWAN using the example provided in this section. However, I'm not using the TTN network, and my gateway doesn't have the option to reset DevNounce. This is why, when signaling a reset or deep sleep of the micro, it's unable to reconnect. Do you have an example where it's not necessary to enable the server to allow previously seen connections? Thank you very much.

lewisxhe, 2025-09-06:
Please see the example here, the application layer is the same
https://github.com/jgromes/RadioLib/tree/master/examples/LoRaWAN

DanielPerezGuzman22, 2025-09-09:
Hello, thank you very much for your response. However, I have already reviewed these examples, and as I understand it, they are the basis for the examples you share on your site for use with the T-Beam. In these examples, it always says:

“Running this example REQUIRES you to check ‘Resets DevNonces’ on your LoRaWAN dashboard. Refer to the network’s documentation on how to do this.”

But in my case, as I mentioned, I am using a RAK server and I am not connected to TTN, so it is impossible for me to perform “Resets DevNonces.” That is why I was asking about an example where it is not necessary to do this. Thank you very much.

DanielPerezGuzman22, 2025-09-10:
Yes, that is the example I use to send data from my T beam supreme, and it works correctly, but when I reset it or apply a deep sleep, it is impossible to connect again, and what I want is to put it into a deep sleep so that the battery lasts much longer.

closed 2025-10-26 by github-actions[bot]