Xinyuan-LilyGO/LilyGo-LoRa-Series #260: T3S3 LR1121 - GPIO17 pin for I2C SLC

vladm, 2025-09-10:
Hi - 

I'm looking through images and the board itself and I can't find GPIO17 as I want to connect to I2C sensor.  There is GPIO18 for Data but can't find scl.   I understand OLED is using those as well, just wanted to plug parallel to OLED.  

Please advise.

lewisxhe, 2025-09-10:
The screen SDA and SCL are not exposed to the outside, you can use the Qwiic interface to connect the sensor

vladm, 2025-09-10:
thank you for your reply. There are 2 connectors 
- one with IO10 and IO21 looks like Lora TX and RX, according to the schematics - so i cant use this
- another is IO43 and IO44 which are available, from what I see and that's the one I should be using?

closed 2025-10-26 by github-actions[bot]